ros-industrial/industrial_calibration #50: error building package without yaml-cpp installed

sebsonf, 2015-04-30:
I have found that the build process will fail with compilation errors if yaml-cpp is not installed on the machine. I guess that dependencies in the CMakeLists.txt and package.xml should be added for the yaml-cpp library to prevent this. 

I have now installed the library manually and added the entry to find_package() in CMakeLists.txt but I still get the following error:

    CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
      Could not find a package configuration file provided by "yaml-cpp" with any
      of the following names:

    yaml-cppConfig.cmake
    yaml-cpp-config.cmake

      Add the installation prefix of "yaml-cpp" to CMAKE_PREFIX_PATH or set
      "yaml-cpp_DIR" to a directory containing one of the above files.  If
      "yaml-cpp" provides a separate development package or SDK, be sure it has
      been installed.
    Call Stack (most recent call first):
      industrial_calibration/industrial_extrinsic_cal/CMakeLists.txt:8 (find_package)

Do you have any advice on how to fix this?

shaun-edwards, 2015-04-30:
`yaml-cpp` is distributed in ROS in a package called `cmake_modules`.  You can install this by:

`sudo apt-get install ros-<distro>-cmake-modules`

gavanderhoorn, 2015-04-30:
See also #14.

gavanderhoorn, 2015-11-12:
The configuration phase should now display proper error messages if `yaml-cpp` isn't installed.

closed 2016-04-14 by drchrislewis