ros-industrial/industrial_calibration #58: Calibration node cannot find transform between frames

akgoins, 2015-09-29:
If the calibration node is launched before the tf frames are available, the calibration node never finds the transform between the camera and the reference frame.  This is due to an error in getPoseFromTF() in ros_transform_interface.cpp.  `now = ros::Time::now()` should be added inside the while loop so that the transform listener will try to look for the latest transform information rather than loop infinitely looking for a transform that never existed on startup.

gavanderhoorn, 2015-09-29:
> `now = ros::Time::now()` should be added [..]

Alternatively, using `ros::Time(0)` should always give you the latest transform available, whereas `ros::Time::now()` will use interpolation and the buffer.

drchrislewis, 2017-03-14:
@akgoins Is this still an issue? I have a vague memory of fixing this.

akgoins, 2017-03-14:
Yes, it looks like it was added [here](https://github.com/ros-industrial/industrial_calibration/blob/indigo-devel/industrial_extrinsic_cal/src/ros_transform_interface.cpp#L35).

closed 2017-03-14 by akgoins