ros-industrial/industrial_calibration #59: Calibration precision

VictorLamoine, 2015-10-21:
Hello,

I watched [Chris Lewis conference](https://vimeo.com/106994711) (really great video!) and I have several questions:
- At 22:00, `how do you tell your calibration is good?`, is there any news on that topic?
- What kind of accuracy did you get when calibrating the Asus camera, sub-milimeter, 5 millimeters, 20? (a rough idea is still better than no info at all)
- Would there be an advantage of calibrating the device using the generated point cloud and not the images? Constructing the target frame over the corner of a cube for example.

I will trying to use this package with the [David SLS-2 sensor](http://www.pointclouds.org/documentation/tutorials/davidsdk.php) that I recently integrated into PCL. I'll make a tutorial about that if I succeed.

Bye

drchrislewis, 2015-10-21:
@VictorLamoine @shaun-edwards 
The accuracy of calibration is determined by several things. The best I can do for you is this:
1. Make sure your residual error is low. In Ceres-solver terminology the is the SSE or sum of squared error. I usually take the square root and divide by the number of observations to get a per-pixel error. 
2. If #1 is true, then the co-variance provides an order of magnitude measure of accuracy.

As for the Asus camera calibration accuracy, and assuming you mean extrinsic calibration, my order of magnitude estimate is 5mm. This accuracy depends heavily on the observation geometry. for example, a larger target provides more accuracy. 

Now for the bad news. The intrinsic calibration you are probably using is in my opinion flawed. Matlab and Opencv use the exact same cost function and distortion model, so it does not matter which you use. They are flawed because they did not check #2 from above. If they had, they would have observed that distortion parameters are correlated to focal length and center point. This explains why the results from intrinsic calibration have such wide variance. For example, the center point (cx,cy) is directly related to residual error. A small change in center point location is equivalent to shifting all the pixels over, and therefore shows up as a corresponding increase in residual error of about the same size. However residual error is typically .25 pixels, while the variance on cx,cy is about 6-10 pixels when using the standard algorithm.  I think I will have an alternate intrinsic calibration procedure soon that should fix this issue. For now, place a cal-target 1 meter from your Asus, calculate its location using the target_finder node, and make sure the distance returned using your intrinsic values is 1meter. Note, the focal point of your camera is someplace behind the lens, and not accurately known, so +-1/4 or 6mm. 

Lastly, it is a bad idea to use the point cloud to calibrate. This point cloud is a product of intrinsic and stereo calibration. One of the pull requests also has the beginnings of a automated point cloud calibration process. We should have a video in the coming months explaining both intrinsic and point cloud calibration.

xiqiaosang, 2015-11-10:
hi

xiqiaosang, 2015-11-10:
hi,i meet a problem when i was doing the industrial_extrinsic_calibration/tutorial/Multi_camera_calibration,i following the tutorial to modify the file adapt to my camera,but result is so puzzled.can you give me some advice@drchrislewis@VictorLamoine @shaun-edwards

sampsonh, 2015-11-11:
hi

drchrislewis, 2015-11-11:
@xiqiaosang 
What problems are you having?

xiqiaosang, 2015-11-12:
@drchrislewis Thank you very much for your kindly reply. My email address is zliu7@e.ntu.edu.sg  The problem is described as follows:
"we want to use two kinect cameras to do the calibration and a circlegrid7x5 target. And the packages we used are: drchrislewis/industrial_calibration and ros-industrial/industrial_calibration_tutorials, both in indigo devel. And we defined the the cameras and target as static.But we got some problems. 

1.Shoule we have to use a robot in calibration? We just put the cameras and target in sedentary postions instead of using a robot to control the target. Is that OK? If that is ok, should we place the actual positon of the cameras or target as close as possible to the initial values in the xacro file?

2.When executing the calibration, one error appears.
[ERROR] [1447295605.851915052]: Calibration ran successfully, but error was larger than allowed by caller
How to solve this error?

3.And using your package to do the calibration, the first camera we use could observe the target, but the second one shows no image and could not observe the same target. But the target initialed in camera_scene_caljob.yaml are the same. And we are sure that the target is within both of the two cameras' views. What's more, this situation won't appear when using the pacakge - ros_industrial/industrial_calibration.

4.Which type of cost_type should we use? In some test yaml file, we found that most of them use FixedCircleTargetCameraReprjErrorPK, but we could not get the exact position and orientation of the cameras when using it."
Regards,
xiqiaosang

gavanderhoorn, 2015-11-12:
@xiqiaosang: could you plase open a new issue for your problems/questions? You can just copy/paste what you wrote in your comment here. Afterwards, remove your comments from this issue.

This thread is really only meant for @VictorLamoine's issue.

xiqiaosang, 2015-11-12:
@gavanderhoorn Sure, it is a good suggestion. :)

closed 2016-04-14 by VictorLamoine