ros-industrial/industrial_calibration #63: calibration result far from "correct"

xiqiaosang, 2015-11-12:
The problem is described as follows:
"we want to use two kinect cameras to do the calibration and a circlegrid7x5 target. And the packages we used are: drchrislewis/industrial_calibration and ros-industrial/industrial_calibration_tutorials, both in indigo devel. And we defined the the cameras and target as static.But we got some problems.

1.Shoule we have to use a robot in calibration? We just put the cameras and target in sedentary postions instead of using a robot to control the target. Is that OK? If that is ok, should we place the actual positon of the cameras or target as close as possible to the initial values in the xacro file?

2.When executing the calibration, one error appears.
[ERROR] [1447295605.851915052]: Calibration ran successfully, but error was larger than allowed by caller
How to solve this error?

3.And using your package to do the calibration, the first camera we use could observe the target, but the second one shows no image and could not observe the same target. But the target initialed in camera_scene_caljob.yaml are the same. And we are sure that the target is within both of the two cameras' views. What's more, this situation won't appear when using the pacakge - ros_industrial/industrial_calibration.

4.Which type of cost_type should we use? In some test yaml file, we found that most of them use FixedCircleTargetCameraReprjErrorPK, but we could not get the exact position and orientation of the cameras when using it."
Regards,
xiqiaosang

xiqiaosang, 2015-11-16:
Dear @drchrislewis :
     Thank you very much for your last reply. Now I followed your instructions, to put two cameras far away .etc. The result now is much better: Camera 1 can observe the target, and the calibration result for camera 1 seems to be accurate. However, camera 2 can not recognize the same circles-target, the target we used is modified_circle_5x5, and the following errors show up:
 
[ERROR] [1447663919.912224744]: intermediate frame from world_frame to world_frame
[ERROR] [1447663919.912477029]: roi = 0 640 0 480
[ERROR] [1447663919.912520319]: roi = 0 640 0 480
[ERROR] [1447663920.291806547]: cameras triggered
[ERROR] [1447663920.292018266]: waiting for camera observations to finish
[ERROR] [1447663920.292073112]: observations done
[ERROR] [1447663920.292141341]: extracting observations
[ERROR] [1447663920.564058955]: intermediate frame pose = 0.000000 0.000000 0.000000  0.000000 0.000000 0.000000
 [ INFO] [1447663920.564388375]: Processing 25 Observations 
[ERROR] [1447663920.564686868]: waiting for camera observations to finish
[ERROR] [1447663920.564804600]: observations done
[ERROR] [1447663920.564922810]: extracting observations
[ERROR] [1447663920.700662482]: couldn't find
[ INFO] [1447663920.701382036]: Processing 0 Observations 

 
And I find out that the final result for circles-target and camera 2 in world frame remains the same as their initial values, no changes. I guess the reason behind this is due to camera 2's failure of recognizing the circles-target. 
How can I solve this problem? Wish to get your opinion.
 
Best regards,
xiqiaosang

drchrislewis, 2015-11-16:
There are several things one can do to get the opencv circle finder to
detect the grid of circles. The easiest is to move the grid so that it is
detected. If you've run ROS's calibration, you will note that when the grid
is too close, too far, or too oblique, or the lighting/exposure is not
good, that the target is not found. We use the exact same subroutine to
find the grid. The second thing you can do is add a region of interest
which narrows the search. Often this works, but I'm not sure why. Next, you
can increase the size of the grid, either more circles, or larger
diameter/spacing.  Next there are lots of parameters for the grid finder. I
use the defaults, but if you modify the code to adjust these you might get
it to find the target. Oh, also when the target is on the verge of being
located, meaning it finds it sometimes, but not always, you can simply add
repeated observations. Have it try 10 times. There is some image noise, so
the images won't be exactly identical. This last idea rarely works.

closed 2016-04-14 by drchrislewis