ros-industrial/industrial_calibration #64: industrial_calibration catkin_make error(hydro)

sampsonh, 2015-11-17:
when i download the industrial_calibration-hydro-devel and compiling,i met the problem
/usr/include/yaml-cpp/node.h:39:3: note:   candidate expects 0 arguments, 1 provided<br>/usr/include/yaml-cpp/node.h:33:21: note: YAML::Node::Node(YAML::Node&)<br>/usr/include/yaml-cpp/node.h:33:21: note:   no known conversion for argument 1 from ‘const YAML::Node’ to ‘YAML::Node&’<br>/home/t420s/moveit/src/industrial_calibration-hydro_devel/industrial_extrinsic_cal/include/industrial_extrinsic_cal/yaml_utils.h: At global scope:<br>/home/t420s/moveit/src/industrial_calibration-hydro_devel/industrial_extrinsic_cal/include/industrial_extrinsic_cal/yaml_utils.h:82:36: error: variable or field ‘parseKeyDValue’ declared void<br>/home/t420s/moveit/src/industrial_calibration-hydro_devel/industrial_extrinsic_cal/include/industrial_extrinsic_cal/yaml_utils.h:82:30: error: ‘const_iterator’ is not a member of ‘YAML’<br>/home/t420s/moveit/src/industrial_calibration-hydro_devel/industrial_extrinsic_cal/include/industrial_extrinsic_cal/yaml_utils.h:82:52: error: ‘it’ was not declared in this scope<br>/home/t420s/moveit/src/industrial_calibration-hydro_devel/industrial_extrinsic_cal/include/industrial_extrinsic_cal/yaml_utils.h:82:68: error: expected primary-expression before ‘&’ token<br>/home/t420s/moveit/src/industrial_calibration-hydro_devel/industrial_extrinsic_cal/include/industrial_extrinsic_cal/yaml_utils.h:82:69: error: ‘key’ was not declared in this scope<br>/home/t420s/moveit/src/industrial_calibration-hydro_devel/industrial_extrinsic_cal/include/industrial_extrinsic_cal/yaml_utils.h:82:74: error: expected primary-expression before ‘double’<br>/home/t420s/moveit/src/industrial_calibration-hydro_devel/industrial_extrinsic_cal/include/industrial_extrinsic_cal/yaml_utils.h: In function ‘bool industrial_extrinsic_cal::yamlNodeFromFileName(std::string, YAML::Node&)’:<br>/home/t420s/moveit/src/industrial_calibration-hydro_devel/industrial_extrinsic_cal/include/industrial_extrinsic_cal/yaml_utils.h:93:12: error: ‘LoadFile’ is not a member of ‘YAML’
is there a problem with the yaml file,can you give me some advice?@drchrislewis

drchrislewis, 2015-11-17:
@sampsonh There should be three yaml_utils h files. Under hydro, typically the old yaml parser works. The new yaml parser has a completely different syntax. That's why we put all the yaml syntax into a single file called yaml_utils.h. All you should have to do is copy either
yaml_utils_hydro.h to yaml_utils.ch or yaml_utils_indigo.h to yaml_utils.h and it should compile.

sampsonh, 2015-11-17:
it works,thanks!!

closed 2016-04-14 by drchrislewis