ros-industrial/industrial_calibration #65: relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object

StanleyYake, 2015-11-19:
Hello there:
    I am using Hydro on Ubuntu 12.04 64 bit, and I have stucked on the `relocation R_X86_64_32 against .rodata.str1.1 can not be used when making a shared object` for couple of days, and still have no good solution, sounds like a platform problem. Here is what I have referenced:http://stackoverflow.com/questions/9637551/relocation-r-x86-64-32s-against-rodata-while-compiling-on-64-bit-platform

[image]

VictorLamoine, 2015-11-19:
Most probably you did not compile gflag, glog, ceres as shared libraries; the error does not seem to berelated to cere though

I've recently updated the instructions
http://wiki.ros.org/industrial_extrinsic_cal#Installation

Try re-compiling and installing the libraries after checking `BUILD_SHARED_LIBS=ON` wherever it exists.

StanleyYake, 2015-11-19:
VictorLamoine, Thanks for reply. I have went to the gflag, glog, ceres directory to do`sudo make uninstall` and `make clean` , then re-compiling and installing as the url you pasted, but still the same problem.

shaun-edwards, 2015-11-19:
The install page for Ceres mentions some problems with SuiteSparse, see [here](http://ceres-solver.org/building.html#linux).

StanleyYake, 2015-11-20:
@shaun-edwards Thank you for your infomation. Yes, I doubt the Ceres not compiled as a shared library, so I re-installed it again after add the PPA, but how can I check it trully installed from the PPA? I am not familiar with this. Here is `apt-cache policy libsuitesparse-dev
`
[image]

it is from the universe, dose it from PPA??

sergiomb2, 2016-01-14:
I suspect  that is hardened builds, looks like, to me, that is a similar problem of http://answers.opencv.org/question/67149/error-building-opencv-300-with-ffmpeg/?answer=83697#post-id-83697

datomnurdin, 2016-02-24:
Any solutions?

cumtchenchang, 2019-12-13:
I also encounter this error, how to solve it?

yangxh31, 2019-12-27:
> I also encounter this error, how to solve it?

i solved this problem as below in cmakefile.txt
set(Boost_USE_STATIC_LIBS OFF)
set(Boost_USE_NULTITHREADED OFF)
set(Boost_USE_STATIC_RUNTIME OFF)

sergiomb2, 2019-12-27:
Relocation R_X86_64_PC32 against undefined symbol , usually happens when LDFLAGS are set with hardening and CFLAGS not .
Common fixes :
https://bugzilla.redhat.com/show_bug.cgi?id=1304277#c3
https://github.com/rpmfusion/lxdream/blob/master/lxdream-0.9.1-implicit.patch

https://stackoverflow.com/a/38579792/778517

drchrislewis, 2019-12-27:
@cumtchenchang @yangxh31 @sergiomb2 I've never seen this error. I'm not sure if you all have "fixed" it. If you think you have please submit a PR. I'll be glad to test it on my machine. I really don't understand the "hardening" or any what any of the set(BoostXX) commands might do.

sergiomb2, 2019-12-28:
@drchrislewis when Fedora starts harden all packages [1], this error appears a lot and when I googled it I found this issue, is not a ROS specific problem.
So when we compile the software and add GCC building flags to hardening,  but if cflags lose the harden flags and ldflags don't ,  we got this error ... 
Or make sure that you add -fPIC to CFLAGS or CXXFLAGS for make-based projects.

[1]
https://fedoraproject.org/wiki/Changes/Harden_All_Packages