ros-industrial/industrial_calibration #66: package **rgbd_depth_correction** build ERROR with catkin

ille-90, 2015-12-09:
Hello guys,

I'm using Ubuntu 14.04 (Trusty) and try to install **rgbd_depth_correction** package with the command: 
**catkin build rgbd_depth_correction**.
I had no problems installing the other packages from the industrial_calibration package  with catkin tools, only **rgbd_depth_correction** fails.

I checked for similar problems but could not find any solution for indigo, just an old package for groovy to install the **bullet** package from http://wiki.ros.org/bullet.
Regarding the source on https://github.com/bulletphysics/bullet3 I followed installation instructions and installed bullet3 from source but that did not solve the problem. 

Here is the Error Code:

```
ille@ille-HP-Pavilion-15-Notebook-PC:~/catkin_ws$ catkin build rgbd_depth_correction 
---------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /home/ille/catkin_ws
Source Space:       [exists] /home/ille/catkin_ws/src
Build Space:        [exists] /home/ille/catkin_ws/build
Devel Space:        [exists] /home/ille/catkin_ws/devel
Install Space:      [exists] /home/ille/catkin_ws/install
DESTDIR:                     None
---------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
---------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
---------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------- 
Found '30' packages in 0.1 seconds. 
Starting ==> kdl_conversions                                                                                                                     
Starting ==> tf2_msgs                                                                                                                            
Finished <== kdl_conversions          [ 0.2 seconds ]                                                                                            
Finished <== tf2_msgs                 [ 0.5 seconds ]                                                                                            
Starting ==> tf2                                                                                                                                 
Finished <== tf2                      [ 0.2 seconds ]                                                                                            
Starting ==> tf2_bullet                                                                                                                          
Starting ==> tf2_py                                                                                                                              
                                                                                                                                                 
[tf2_bullet] ==> '/home/ille/catkin_ws/build/tf2_bullet/build_env.sh /usr/bin/cmake /home/ille/catkin_ws/src/geometry_experimental/tf2_bullet -DCATKIN_DEVEL_PREFIX=/home/ille/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/ille/catkin_ws/install' in '/home/ille/catkin_ws/build/tf2_bullet'
-- checking for module 'bullet'
--   package 'bullet' not found
CMake Error at /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:283 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:337 (_pkg_check_modules_internal)
  CMakeLists.txt:7 (pkg_check_modules)

-- Using CATKIN_DEVEL_PREFIX: /home/ille/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/ille/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/ille/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ille/catkin_ws/build/tf2_bullet/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- Using these message generators: gencpp;genlisp;genpy
-- Configuring incomplete, errors occurred!
See also "/home/ille/catkin_ws/build/tf2_bullet/CMakeFiles/CMakeOutput.log".
See also "/home/ille/catkin_ws/build/tf2_bullet/CMakeFiles/CMakeError.log".
[tf2_bullet] <== '/home/ille/catkin_ws/build/tf2_bullet/build_env.sh /usr/bin/cmake /home/ille/catkin_ws/src/geometry_experimental/tf2_bullet -DCATKIN_DEVEL_PREFIX=/home/ille/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/ille/catkin_ws/install' failed with return code '1'
                                                                                                                                                  
Failed   <== tf2_bullet               [ 0.8 seconds ]                                                                                            
Finished <== tf2_py                   [ 0.9 seconds ]                                                                                            
[build] There were '1' errors:                                                                                                                   

Failed to build package 'tf2_bullet' because the following command:

# Command to reproduce:
cd /home/ille/catkin_ws/build/tf2_bullet && /home/ille/catkin_ws/build/tf2_bullet/build_env.sh /usr/bin/cmake /home/ille/catkin_ws/src/geometry_experimental/tf2_bullet -DCATKIN_DEVEL_PREFIX=/home/ille/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/ille/catkin_ws/install; cd -

# Path to log:
cat /home/ille/catkin_ws/build/build_logs/tf2_bullet.log

Exited with return code: 1 
                                                                                                                                                 
[build] Runtime: 1.8 seconds 
ille@ille-HP-Pavilion-15-Notebook-PC:~/catkin_ws$ 
```

gavanderhoorn, 2015-12-09:
Looking at this:
```shell
Starting ==> tf2_bullet                                                                                                                          
Starting ==> tf2_py                                                                                                                              

[tf2_bullet] ==> '/home/ille/catkin_ws/build/tf2_bullet/build_env.sh /usr/bin/cmake /home/ille/catkin_ws/src/geometry_experimental/tf2_bullet -DCATKIN_DEVEL_PREFIX=/home/ille/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/ille/catkin_ws/install' in '/home/ille/catkin_ws/build/tf2_bullet'
-- checking for module 'bullet'
--   package 'bullet' not found
CMake Error at /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:283 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:337 (_pkg_check_modules_internal)
  CMakeLists.txt:7 (pkg_check_modules)
```
I'd say this is actually an issue with `tf2`, and not with this package. If I'm correct it would be [this](https://github.com/ros/geometry_experimental/blob/c4d7465591456a8cc1148ef34f6170cc2f3153ef/tf2_bullet/CMakeLists.txt#L6-L7) line in `CMakeLists.txt` of `tf2_bullet` that is what runs into that error.

How have you setup your build environment? Can you list the steps you took before running `catkin build`?

gavanderhoorn, 2015-12-09:
As for the `bullet` dependency itself, that actually maps onto `libbullet-dev` on Ubuntu Trusty (from [here](https://github.com/ros/rosdistro/blob/b979cf3547a6139c134961c7f2d3b033549c6fd6/rosdep/base.yaml#L348-L383)):

```yaml
bullet:
  [..]
  trusty:
    apt:
      packages: [libbullet-dev]
```

Running `rosdep` should have resolved that for you.

ille-90, 2015-12-09:
found this one here http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-indigo-tf2-bullet/
Installed the package with command
```
sudo apt-get install ros-indigo-tf2-bullet
```
After that
```
catkin build tf2_bullet
```
was successful.

closed 2015-12-09 by ille-90

gavanderhoorn, 2015-12-10:
> found this one [here](http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-indigo-tf2-bullet/) Installed the package with command
```
sudo apt-get install ros-indigo-tf2-bullet
```

Yes, that is what `rosdep` should have done for you. Did you run `rosdep {check,install} --from-paths /path/to/your/ind_cal_ws/src --ignore-src` before running `catkin build`?

> After that
```
catkin build tf2_bullet
```
was successful.

Unless I'm misunderstanding something, you don't need to build binary pkgs using `catkin`. That's the whole point: they've already been compiled. If you have a source checkout of `tf2` in your workspace, that's a different (dependency) problem, but that is what `rosdep` should've sorted out for you.