ros-industrial/industrial_calibration #70: Extrinsic calibration deprecated?

schelian, 2016-01-16:
Hi all, the tutorial on extrinsic calibration (http://wiki.ros.org/industrial_extrinsic_cal/Tutorials/Camera%20to%20Target) is deprecated.  It refers to files that do not exist (e.g., cell_calibration.launch, world_to_target_tf.launch, etc.).

May I know how people are doing their extrinsic calibration these days?  (Sorry if this question is that of a n00b.)

VictorLamoine, 2016-04-14:
The extrinsic calibration is not deprecated but the tutorials are out-dated.
You can look at [godel](https://github.com/ros-industrial-consortium/godel) project for an example of how the calibration is done.

A good start is looking at [godel_robots](https://github.com/ros-industrial-consortium/godel/tree/indigo-devel/godel_robots).

drchrislewis, 2017-03-14:
@AustinDeric There is a working example of using the kinetic-devel branch of both
industrial_calibration_tutorials and industrial_calibration
"roslaunch ind_cal_multi_camera calibrate_from_images.launch" 
We will not likely get to a line by line procedure for calibrating a camera to the robot its mounted on because that involves setting up lots of stuff including moveit with appropriate hooks for modifying the transforms being calibrated.

drchrislewis, 2017-03-22:
@schelian There is a simple working example under industrial_calibration_tutorials that runs under the kinetic-devel branch(calibrate_from_images.launch). Other tutorials may also be made to run, but require modifications.  These examples are used in training exercises and only updated as necessary for each class to match the hardware and current state of the library.

closed 2017-03-22 by drchrislewis