ros-industrial/industrial_calibration #82: dependence Error in indigo-devel branch

RyodoTanaka, 2016-06-02:
Hi, thank you for this awesome package.
I tried to build in indigo-devel branch. And occurred Error.
Before `catkin_make`, I installed all of `gflags`, `glog`, and `Ceres`, following to [ROS industrial_extrinsic_cal wiki](http://wiki.ros.org/industrial_extrinsic_cal).
Then I tried rosdep and occurred following error.
```bash
cd <catkin_ws>
rosdep install -i -y --from-paths src

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
industrial_extrinsic_cal: Cannot locate rosdep definition for [camera_aravis]

```
I found the `camera_aravis` is not supported in indigo.
So, I commented out that dependence. 
And `rosdep` again, it works.
Of course, there is no error I do `catkin_make`.
Is it OK, only comment out that dependence ?

Thank you for your regards.

VictorLamoine, 2016-06-02:
If it still builds on Travis, then I'm pretty sure it's safe to remove it :)

RyodoTanaka, 2016-06-02:
I removed `camera_aravis` dependence at PR #83 .

gavanderhoorn, 2016-06-02:
`camera_aravis` was/is used in some of the launch files. The fact that Travis doesn't fail the build when you remove the dependency doesn't really mean much, unless the CI build also runs some `roslaunch` tests (which afaik, it doesn't).

@drchrislewis should know whether the aravis dependency is still needed / wanted.

VictorLamoine, 2016-06-02:
Ah right, I just disabled the tests!

drchrislewis, 2016-06-02:
camera_aravis is only required to run the launch files. We should remove it from the package.xml
I use it for working with Basler cameras. You can get a version that works on indigo from
https://github.com/drchrislewis/camera_aravis.git
It is forked from takiaine/camera_aravis, with a couple of minor changes. There are at least three different camera_aravis stacks out there, none of which work for me without some edits. I've had a pull request in for some time, but no response, so perhaps its a stale stack. 
Please let me know if you have any questions in using the calibration stacks.

RyodoTanaka, 2016-06-02:
@VictorLamoine @drchrislewis 

Ah, I understand.
Thank you for your advice.

closed 2016-06-02 by RyodoTanaka