ros-industrial/industrial_calibration #87: Extrinsics calibration instructions

JimmyDaSilva, 2016-09-12:
Hello,

I have been trying to use the the `industrial_calibration` package on indigo to calibrate a camera mounted on my robot with a fixed checkerboard. I am also using the `industrial_calibration_tutorials` package and the mutable joints to see the results as I calibrate.

When I calibrate the `target_frame` successfully gets to the right position on the image. 
[image]

But if I move the robot around I can see that there is a shift in the calibration.
[image]

The calibration procedure only takes one image with the checkerboard to calibrate which seems enough to get a solution, but not enough to get precise results.
I couldn't find a way to calibrate using several acquisitions yet ... Also the deprecated tutorials doesn't mention how to calibrate a mounted camera with a static target.

I also tried recalibrating the intrinsics a couple of time, but in the end I always get a shift on my calibration.
Does any one have a some experience with a similar problem?

Any help will be sincerely appreciated!

Here are the files I am using :
https://github.com/JimmyDaSilva/industrial_calibration_tutorials/tree/test_checkerboard/ind_cal_multi_camera

drchrislewis, 2016-09-12:
Jimmy, 
    First, one must use more than one image to perform the calibration. The reason is a bit subtle since each checkerboard intersection point provides two equations and we only have 12 unknowns (xyzrpy of both the camera to robot and target to robot). However, these equations turn out to be dependent unless you take several images. Therefore take multiple images from multiple perspectives. In addition, I suggest you use the modified circle grid instead of a checkerboard. The reason is that the orientation of the modified circle grid is never ambiguous. The origin circle of this grid is larger than the rest. 
As for a mounted camera with a static target, is the target's location known exactly relative to the robot? If so, then one can swap out the cost function and perform a different calibration. I'd have to look through the existing cost functions to see if one exists for that. If not, it is very straight forward to write a new one. All the calibration cost functions have a similar form. The constructor defines all the known quantities necessary, and the evaluate function computes the residual. Using the current value of the parameters, it first projects the target points into the camera's coordinate frame and then into the image. The residual is the difference between the observation of the point and that predicted in the projection. The hard part is understanding the syntax. The really cool part is that Ceres computes the Jacobian analytically directly from your function code. Still, I'd suggest you get existing calibration working before tackling a new cost function.

JimmyDaSilva, 2016-09-12:
@drchrislewis, thank you for your answers.

1) That's what I thought, I suspected that several perspectives are need for calibrating with a board.
My biggest problem is that I don't know how. Everytime I call the calibration service it starts over without take the previous observations into account.
Could you explain how I should set the params to be able to use several images before calibration is run?

2) Yes the target is precisely placed on the robot's table and I am publishing its static transform.

drchrislewis, 2016-09-12:
@JimmyDaSilva just put multiple scenes in your caljob.yaml file. It collects images from every scene and creates a SSE cost over all scenes and all observations in those scenes. In each scene, the robot moves to a different location. The thing that tells the robot to move is the scene trigger. Define one that moves the robot to a new location, but one where the camera can see the target. This whole thing is meant to allow one to define a calibration procedure, and then run it exactly the same way any time a camera gets bumped. It takes a fair bit of work to set up the first time. Please ask as many questions as necessary. There is a caljob creator package that can help create the scenes. I recall one just moves the robot around via its teach pendant, and it creates a new scene by reading the joint values. I think one had to set some ros parameter to true each time one wanted another scene. Once you get it all running, please consider checking the covariance of the solution(Ceres can return that). That's the only way to be certain one has enough diversity in observations of any calibration.

JimmyDaSilva, 2016-09-12:
@drchrislewis Thank you so much! Will try that tomorrow.
For now I am having trouble using the modified circle grid. I get the following:
`couldn't find 5x5 modified circle target in /head_cam/image_rect_color, found only 0 pts`
[image]

drchrislewis, 2016-09-12:
@JimmyDaSilva the grid appears to be well lit and completely visible. We use open-cv's grid finder. Sometimes it is finicky, but the only other thing I can think of is if you defined a ROI the excluded some points. Make sure the roi is the same size as the whole image or at least includes all the circles.

JimmyDaSilva, 2016-09-12:
@drchrislewis The caljob_creator is great.
I am still stuck with the modified circle target. Weird because simple circle detection works. I printed out this one but no more success:
https://github.com/ros-industrial/industrial_calibration/blob/indigo-devel/industrial_extrinsic_cal/targets/mcircle_10x10_dia12p5_sp20.pdf
[image]

drchrislewis, 2016-09-12:
@JimmyDaSilva When you switched from a checkerboard to an m-circle did you change your target.yaml file to reflect both the size (5x5) and the type change? target type is 2 for m-circle. This is not clear, and should be fixed to have named types instead of an integer matching some dot-h file. Types are defined in ros_camera_observer.h.

JimmyDaSilva, 2016-09-12:
@drchrislewis yaml files seem to be fine.
I did change the target_name, target_type and so on ... Proof is that I get an error mentioning modified the circle grid:
`couldn't find 10x10 modified circle target in /head_cam/image_rect, found only 0 pts`

Also I add some debugging lines in the camera_observer node and it really seems that the opencv fonction fails for ModifiedCircleGrid, at least in my case.

drchrislewis, 2016-09-12:
@JimmyDaSilva sometimes defining a ROI helps. Sometimes, for high res cameras, the blobs get too big for the defaults of the simple blob detector. Try moving the robot back away from the target a bit.

drchrislewis, 2016-09-12:
@JimmyDaSilva Also, I typically define 20-30 robot poses. When some don't capture the target, for whatever reason, it still runs just fine using data from the rest of the scenes. You can manually delete scenes that consistently fail to speed the process.

JimmyDaSilva, 2016-09-13:
Dear @drchrislewis thank you for the huge help. Just moving far from the target worked
[image]

Now I just need to use the caljob_creator to use several images for calibration.

I know it takes quite some time to document things, but I think people would love an update of the tutorials !

Thanks again for your time.

JimmyDaSilva, 2016-09-13:
@drchrislewis Sorry Chris, I have added 11 scenes but in the end it cannot find a solution. The solution is found on the other side of the table. Error is:
```
[ INFO] [1473760800.577366849]: Calibration Sucessful. Initial cost per observation = 330719.267894 final cost per observation 15300.166004
[ERROR] [1473760800.577410068]: Calibration ran successfully, but error was larger than allowed by caller
```

[image]

JimmyDaSilva, 2016-09-13:
The `cost_type` autogenerated by the caljob_creator was not the one I needed!
Works really good.
Thanks again. Awesome work on this package !

JimmyDaSilva, 2016-09-13:
@drchrislewis I am kind of lost with the cost functions and don't really know which one to use. To me the name are not self explanatory.
So far I think `FixedCircleTargetCameraReprjErrorPK` is the one which gave me the best solution. But the final cost is still of 22

JimmyDaSilva, 2016-09-13:
Also in the camera file, should I use the values of K or P to fill in the focals if I use the `image_rect` ? What about the distortions values, set to the ones in D, or 0 ? I am confused.

drchrislewis, 2016-09-13:
In general, when doing extrinsic calibration, subscribe to the rectified image. I thought we fixed the code so that the camera observer got the focal length and the center point info from the camera publisher. 
Next, the cost functions are a mess. Many were used once, in ways that were never documented, and should therefore be deleted. The best thing to do is write down what you know first, all of these values should be in the constructor. What you are trying to adjust will be in the evaluate() function. The syntax of overloading the () operator hides the evaluate function for people like me. Here is what I think you know:
1. the observation ob_x, ob_y
2. Where that point is relative to the target frame. Point3d point
3. Where the link upon which the camera is mounted was when the observation was taken, Pose6d link_pose
4. camera intrinsic parameters fx,fy,cx,cy for pinhole model
Now, what you want to find out:
1. transform from link to camera frame. a six vector containing x,y,z,angle-axis 
2. transform from target to robot base or world, again a six vector
Now, there are still lots of candidates
1. LinkCameraTargetReprjErrorPK()
2. LinkCircleTargetCameraReprjErrorPK()
3. PosedTargetCameraReprjErrorPK()
4. LinkCameraCircleTargetReprjErrorPK()

Then you have to see which one does what you want. Notice all say PK, the points are known in the target frame. This is analogous to having a calibrated camera, the target is calibrated so that point locations are known. 
You used a circle target with known circle diameters, but it is not generally necessary to use this information. There is a slight bias when circles are viewed from an angle that depends on the size of the cirlce, but is generally less than a few hundredths of a pixel. You may or may not want to use a cost function that calls cameraCircResidual() instead of cameraPntResidual(). 
The cost function you want will do the following order of operations
1. project the point into the world/robot coordinates using the unknown target 6 vector
2. project the point into the link coordinates using the known link transform (inverted)
3. project the point into the camera frame using the unknown camera 6 vector
4. compute the residual
At first glance, I think 1,2 or 4 might work. It looks like 4 does not use the c_dia because its commented out and replaced with the simpler code...oops. 
I would suggest LinkCameraCircleTargetReprjErrorPK() anyway.

JimmyDaSilva, 2016-09-13:
I've tried all of the candidates, with and without `cameraCircResidual()`. Error is even bigger.
My best results are still with `FixedCircleTargetCameraReprjErrorPK` and `cameraCircResidual()` but it's still off.

I am trying to generate a pattern with a smaller size precisely. Do you know what is the rule for the size of the biggest circle ?

drchrislewis, 2016-09-13:
If your kinematics parameters are not that accurate, or your printer is not that accurate, 22 pixels may be an ok result. For the images I saw, I would doubt the circle-residual would have much affect compared to all the other potential errors. Be very sure your views have lots of different perspectives. Don't just move around in a plane. When in doubt, check the covariance.

closed 2016-10-31 by JimmyDaSilva

Sinchiguano, 2019-01-30:
Hi,

Just a question, do you think it is possible to find the extrinsic parameter for the following setup, one target (circular target) and one camera. if so, could you tell  me in what way to do it. Please, i am quite new with calibration thing.

Best regards!
Cesar

drchrislewis, 2019-01-31:
@Sinchiguano 
With a circle grid target (not just one circle), one can do extrinsic calibration. Look at the localization node. It does just that. Given a set of points on a plane with known locations in that plane, and the location of that plane relative to a camera, one can predict where those points appear in the image. We can also see where they did actually appear in the image. Calibration simply adjust the transform between the camera and the target until the sum of squared observation error is minimized.