ros-industrial/industrial_calibration #88: Build Errors

jshepley14, 2016-11-05:
Hello,
     I am a member of Columbia's ROAM lab run by Dr. Matei Ciocarlie.  Unfortunately, I have not gotten a chance to use the industrial_calibration (indigo-devel) package because the code simply will not build.  There seems to be errors with certain OpenCV variables and functions, specifically having to do with circle_detector. I have managed to debug some of the errors myself, but I am surprised that there are build errors in the first place.  Is there a specific version of OpenCV required?  I am using OpenCV 2.4.  Any other suggestions related to solving the build errors would be appreciated too.
[image]

Thanks,
Joseph

gavanderhoorn, 2016-11-06:
Thanks for the report.

slightly off-topic, but could I please ask you to not use screenshots for what is essentially plain text? You should be able to just copy-paste the terminal text. Be sure to use backticks to format it properly in your issue.

gavanderhoorn, 2016-11-06:
Could you please add some more information to your OP? Which OS is this, which version, how did you install OpenCV, etc.

> [..] I am surprised that there are build errors in the first place.

May I ask why? Software is never frozen in time, especially not when you depend on libraries that are not under your control. We're always tracking a moving target wrt our dependencies.

Building things from sources (as you are doing) means that things like this can happen.

drchrislewis, 2016-11-06:
I know this won't be much help, but my opencv version is 2.4.8. I compile without errors. In addition the class circledetector() was in a beta version, and I put a modified copy of it in our repo. The main distinction was that I fit an ellipse to the contours of blobs rather then just finding their center of mass to find the center of the circle. I saw a slight improvement in accuracy with a circle detector, but wasn't sure if the significance was real. The only reason we wanted to be able to switch between two different feature detectors was for accuracy comparison. We could just pick one or the other and avoid the problem, but I still don't understand why there is a problem assigning a pointer to a base class to point to the derived class. findCirclesGrid() requires a Ptr<FeatureDetector> and circle_detector_ptr_ is a  cv::Ptr<cv::FeatureDetector>  while we have the blob detector declared as: class SimpleBlobDetector : public FeatureDetector. Perhaps there is a difference between our compilers?

jshepley14, 2016-11-15:
Hi all, thank you for your replies.   I am using Ubuntu 14.04.5. As for C++ compilers, they are listed as
g++  4:4.8.2-1ubuntu6       amd64   
and   
g++-4.8   4.8.4-2ubuntu1~14.04.3  amd64
As for OpenCV, I was not the one to install it so I don't know exactly how it was installed, but again, the version is 2.4.8.  I will try to find out how it was installed if possible.  I recognize this is a development branch, and I appreciate your efforts.  

Thanks,
Joseph

VictorLamoine, 2016-11-15:
The `/usr/local/include` path of you OpenCV includes indicates that the OpenCV library used to compile (the one found by CMake) was manually compiled/installed on the machine.

A first test would be to modify the CMake OpenCV path in your catkin workspace, point it to the binary/apt packages and try to compile again. Steps:
- Find the system OpenCV CMake file (something like `find / -iname "*opencv*.cmake` should help)
You should have at least two of them, one in `/usr/include` and one in `/usr/local/include`
- Force CMake in your catkin workspace to use the one in `/usr/include` rather than `/usr/local/include`

If my instructions are not clear, don't hesitate to ask for help!

AustinDeric, 2017-03-03:
@jshepley14 did this work out for you?  Do we need a code fix for this?  if not, i recommend that we close this issue.

In the future, if not a code problem then we should address these issues at answers.ros.org

Thanks everyone for your support!

AustinDeric, 2017-03-07:
@jshepley14 Since we can't duplicate this error i am going to close.  If you continue to experience build errors place see answers.ros.org.

closed 2017-03-07 by AustinDeric