ros-industrial/industrial_calibration #89: OpenCV 3 support

VictorLamoine, 2016-11-10:
Kinetic ships with OpenCV 3 and `industrial_calibration` expects OpenCV 2, this makes CMake fails when checking OpenCV dependency.

At the moment I just replaced
```
find_package(OpenCV 2 REQUIRED)
```
with
```
find_package(OpenCV REQUIRED)
```

In the two CMake files requiring OpenCV

edgeded, 2017-01-03:
Does this work for you? i.e. were you able to build all sources?
In my case CMake could find OpenCV 3 but fails building e.g. cirlce_detector.hpp 

```
[ 85%] Building CXX object industrial_calibration/industrial_extrinsic_cal/CMakeFiles/industrial_extrinsic_cal.dir/src/circle_detector.cpp.o
In file included from .../industrial_calibration/industrial_extrinsic_cal/src/circle_detector.cpp:50:0:
.../industrial_calibration/industrial_extrinsic_cal/include/industrial_extrinsic_cal/circle_detector.hpp:101:46: error: ‘vector’ has not been declared
   virtual void detectImpl( const Mat& image, vector<KeyPoint>& keypoints, const Mat& mask=Mat() ) const;
```

I suppose this is due to different declarations in OpenCV 2 and OpenCV 3

VictorLamoine, 2017-01-03:
Yes I was able to build the sources by de-activating one or two features. I'll get back to you in an hour with a pull request to show you the changes!

edgeded, 2017-01-03:
This would be great! 
I tried some changes by myself but somehow I ended up in a mess.

VictorLamoine, 2017-01-03:
Better late than never.. sorry!

https://github.com/ros-industrial/industrial_calibration/pull/90

edgeded, 2017-01-03:
Unfortunately it doesn't work for me on Ubuntu 16.10
I cloned your https://github.com/InstitutMaupertuis/industrial_calibration but get building errors e.g.

```
.../src/industrial_calibration/industrial_extrinsic_cal/src/nodes/ros_robot_scene_trigger_action_server.cpp:22:
/opt/ros/kinetic/include/moveit/macros/declare_ptr.h:52:16: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   typedef std::shared_ptr<Type> Name##Ptr;       
```

but thank you very much though!

VictorLamoine, 2017-01-03:
Ah, we need C++11 enabled. I'm using gcc 6 so C++14 is enabled by default, that's why I don't see this error.
I'll add that in the pull request right now.

edgeded, 2017-01-03:
Thank you very much!

VictorLamoine, 2017-02-15:
I tried used the calibration procedure again and now I get this error:

```bash
OpenCV Error: The function/feature is not implemented () in detectAndCompute, file /tmp/binarydeb/ros-kinetic-opencv3-3.1.0/modules/features2d/src/feature2d.cpp, line 144
[ERROR] [1487148390.729431750]: Exception thrown while processing service call: /tmp/binarydeb/ros-kinetic-opencv3-3.1.0/modules/features2d/src/feature2d.cpp:144: error: (-213)  in function detectAndCompute

[ERROR] [1487148390.729564226]: Service call failed: service [/TargetLocateService] responded with an error: /tmp/binarydeb/ros-kinetic-opencv3-3.1.0/modules/features2d/src/feature2d.cpp:144: error: (-213)  in function detectAndCompute
```

VictorLamoine, 2017-02-15:
Problem is here:
https://github.com/ros-industrial/industrial_calibration/blob/indigo-devel/industrial_extrinsic_cal/src/ros_camera_observer.cpp#L103

Solution is explained here:
http://stackoverflow.com/questions/30622304/opencv-3-blobdetection-the-function-feature-is-not-implemented-in-detectand

Not sure how I should modify the custom CircleDetector class.

drchrislewis, 2017-03-22:
I believe this is now fixed with retry_dynreconfig PR