ros-industrial/industrial_calibration #95: Laser line / profilometer extrinsic calibration

VictorLamoine, 2017-03-03:
We have found this paper:
http://www.mdpi.com/1424-8220/9/9/7374

- I tried to test the [Keyence experimental driver](https://github.com/ros-industrial/keyence_experimental) but the TCP/UDP ports filtered in our network, I'm waiting for them to be open. I should be able to use the driver next week.

- I'm waiting for a 3D printed interface to mount the sensor on the robot. I should get this next week.

- We are thinking about the target geometry, we're not sure what to use.

From what I have understood we need to be able to build a (3D) frame consistently and precisely on the calibration target whatever the  point of view. Given the fact that we only have a line this is quite difficult to think of.

Rather than using a single line acquisition, I thought we might get a bunch of them with a linear sweeping movement. This way we can get a point cloud that we would store in the first frame user frame and built a 3D user frame from here.

We could for example scan 3 faces of a cube, the problem would be that because the dimensions are the same everywhere we cannot determine the orientation of the cube; we would need markers on the faces to determine which face is which face.

@ AustinDeric @drchrislewis your opinion would be much appreciated

ClayFlannigan, 2017-03-03:
Consider sweeping the laser across a 3-sphere calibration target.  Spherical models can be fit to the resulting point cloud yielding the three sphere centers which can be used to uniquely define a frame.  For example:

[http://www.precisionballs.com/all_ball_bar.php#b3km](http://www.precisionballs.com/all_ball_bar.php#b3km)

Note that you would likely have to paint the spheres to get decent returns.

The SICK Ranger products use methods similar to the referenced paper:

[http://spectronet.de/story_docs/vortraege_2009/091103_vision/091104_1530_nilsson_sick.pdf](http://spectronet.de/story_docs/vortraege_2009/091103_vision/091104_1530_nilsson_sick.pdf)

VictorLamoine, 2017-03-03:
Awesome information Clay thank you!

The three sphere method really looks good; however I don't think we wan't equally spaced spheres otherwise we cannot determine from which point of view the set of sphere was observed.

With a non isosceles triangle there should be no ambiguity.

drchrislewis, 2017-03-03:
I believe the Keyence sensor returns both range and intensity. If that is the case, I see no reason one could not obtain an excellent extrinsic calibration from a planar target (possibly checkerboard, with slight modifications). It is a straight forward exercise given the relative pose between the target and the laser to predict the locations of the transitions from white to black. If one collected a sufficient number of different  relative poses, the pose of the target relative to the base of the robot holding the Keyence and the pose between the wrist and the Keyence should be fully constrained. I would design a relatively small checkerboard target with a unique border on each of the four sides such that all observations of the target included at least one border. Each intensity transition has an (x,y,z) coordinate where y=0 in every measurement and every prediction. The cost would be the differences between x_predict-x_observed and z_predict-z_observed. We could simulate the process to make sure it worked prior to implementation. There is the slight possibility that no matter how inventive we are in diversifying our observations that the optimization is under-constrained. This would be evident in the Jacobian which Ceres-solver computes. I do not have enough geometric intuition to be confident that this would not be the case, but it seems unlikely to me. Every measurement set has an inherent degree of freedom about the axis where the plane of the laser meets the plane of the target. Additional measurements along different planes should tie that axis down.

drchrislewis, 2017-03-03:
@VictorLamoine please see my last comment

AustinDeric, 2017-03-03:
@VictorLamoine We have a slack for these awesome detailed technical discussions.  I find that github issues are not the best place for them.  Can you email me your preferred email address and i will send you an invite.  my email is austin.deric@swri.org.

you too @gavanderhoorn!

VictorLamoine, 2017-03-20:
We will be trying with a ping pong ball today and tomorrow and see what we can get.

Jmeyer1292, 2017-05-04:
Hey @VictorLamoine, it's been forever but I'm finally at a place where I want to start helping out with this laser calibration issue. Can you give me a feel for any progress that you've managed to make?

VictorLamoine, 2017-05-05:
Hey, we have a procedure that has given us great results :+1: 
Right now it's a matlab code.

The procedure uses a calibrated ball (we also tested with a ping pong ball). One has to know precisely the diameter of the object, and it has to be as round as possible.

I'm now familiar with the ROS Keyence driver.

We need to write a ROS package to:
- Move the robot at desired poses
- Fetch and store data from the acquisition
- Process the data and output the calibration matrix

I want to keep these three modules independent to allow easy simulation of the process.
I'm not sure how to integrate this in the current `industrial_calibration` package.

I suggest creating a new package first and see what we can get.
Monday is a national holiday in France so I'll get back to you Tuesday with more details :)

Have a nice week-end!

Jmeyer1292, 2017-05-05:
I'm absolutely on board, especially with respect to the separate modules. 

I wrote a simple touch-point calibration routine for TCP teaching here: https://github.com/Jmeyer1292/tool_point_calibration

I tried to follow a similar model.

Is it possible for you to share this Matlab code?

Jmeyer1292, 2017-05-09:
ping @VictorLamoine

VictorLamoine, 2017-05-10:
I'll send you an email with the code and some details in a few hours!

VictorLamoine, 2017-05-10:
I've sent you and email (and to Christopher Lewis too), ping me if you didn't get if. It was from my pro email (`v*.l*@institutmaupertuis.fr`)

prajval10, 2019-04-05:
Hi @VictorLamoine, we are interested in this issue as well. Is the matlab code mentioned in https://github.com/ros-industrial/industrial_calibration/issues/95#issuecomment-299499938 available on a public repository? Thanks!

VictorLamoine, 2019-04-08:
I have sent you an email with the code (at your GitHub profile email), you should be looking at these repos first:
- https://github.com/mikeferguson/robot_calibration
- https://github.com/Jmeyer1292/robot_cal_tools

I've not worked on this topic since a (very) long time but I still hope I will get the time someday to finish what I have started :)

prajval10, 2019-04-08:
> I have sent you an email with the code (at your GitHub profile email)

Got it! Thanks :)

debunt, 2019-07-04:
Dear @VictorLamoine, could you please either share with me your Matlab code for 2D profile laser? You mentioned about it in [#95 (comment)](url). I try to calibrate by means of a sphere, but I have some trouble. Thank you in advance

gitbagg, 2020-01-31:
Adding another point of interest in this topic. Thanks guys!

udopa, 2020-08-11:
Hi @VictorLamoine, I'm also working with a laserscanner and face the same problem as you with calibration. Could you please share the code with me via Email/Repo? Thanks

VictorLamoine, 2020-08-11:
@udopa I've sent an email at the one publicly available on your GitHub account!

jaredbaggen, 2021-03-02:
@VictorLamoine, would you be open to sharing your procedure? I too put together a non-production version that works well, but is not as accurate as I'd like. Mine is essentially a standard touch-point calibration procedure where the point of my "tool" is the estimated center of the sphere _when_ the diameter of the sphere ~ equals the actual sphere diameter. There is some error though that I'm not accounting for.

VictorLamoine, 2021-03-02:
Sorry, I changed work and I cannot share this data anymore!