ros-industrial/industrial_calibration #123: industrial_extrinsic_cal Build ERROR

Bwaki, 2018-05-23:
Hi, 

I try to compile the pgk industrial_calibration but i got this error (see under), i have try to install reinstall yaml-cpp but no change, i am lost i don't know how to solver this.

If someone have already kick this one can explain me how to resolve it.

Thanks you,

Bwaki
```
CMakeFiles/mutable_joint_state_publisher.dir/src/nodes/mutable_joint_state_publisher.cpp.o: In function `YAML::BadConversion::BadConversion(YAML::Mark const&)':
mutable_joint_state_publisher.cpp:(.text._ZN4YAML13BadConversionC2ERKNS_4MarkE[_ZN4YAML13BadConversionC5ERKNS_4MarkE]+0x7b): undefined reference to `vtable for YAML::BadConversion'
CMakeFiles/mutable_joint_state_publisher.dir/src/nodes/mutable_joint_state_publisher.cpp.o: In function `YAML::TypedBadConversion<double>::~TypedBadConversion()':
mutable_joint_state_publisher.cpp:(.text._ZN4YAML18TypedBadConversionIdED2Ev[_ZN4YAML18TypedBadConversionIdED5Ev]+0x18): undefined reference to `YAML::BadConversion::~BadConversion()'
CMakeFiles/mutable_joint_state_publisher.dir/src/nodes/mutable_joint_state_publisher.cpp.o: In function `YAML::TypedBadConversion<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::~TypedBadConversion()':
mutable_joint_state_publisher.cpp:(.text._ZN4YAML18TypedBadConversionINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEED2Ev[_ZN4YAML18TypedBadConversionINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEED5Ev]+0x18): undefined reference to `YAML::BadConversion::~BadConversion()'
CMakeFiles/mutable_joint_state_publisher.dir/src/nodes/mutable_joint_state_publisher.cpp.o:(.rodata._ZTIN4YAML18TypedBadConversionIdEE[_ZTIN4YAML18TypedBadConversionIdEE]+0x8): undefined reference to `typeinfo for YAML::BadConversion'
CMakeFiles/mutable_joint_state_publisher.dir/src/nodes/mutable_joint_state_publisher.cpp.o:(.rodata._ZTIN4YAML18TypedBadConversionINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEE[_ZTIN4YAML18TypedBadConversionINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEE]+0x8): undefined reference to `typeinfo for YAML::BadConversion'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/bwaki/catkin_ws/devel/.private/industrial_extrinsic_cal/lib/industrial_extrinsic_cal/mutable_joint_state_publisher] Error 1
make[1]: *** [CMakeFiles/mutable_joint_state_publisher.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/bwaki/catkin_ws/devel/.private/industrial_extrinsic_cal/lib/libindustrial_extrinsic_cal.so: undefined reference to `vtable for YAML::BadConversion'
/home/bwaki/catkin_ws/devel/.private/industrial_extrinsic_cal/lib/libindustrial_extrinsic_cal.so: undefined reference to `cv::Feature2D::getDefaultName() const'
/home/bwaki/catkin_ws/devel/.private/industrial_extrinsic_cal/lib/libindustrial_extrinsic_cal.so: undefined reference to `typeinfo for YAML::BadConversion'
/home/bwaki/catkin_ws/devel/.private/industrial_extrinsic_cal/lib/libindustrial_extrinsic_cal.so: undefined reference to `YAML::BadConversion::~BadConversion()'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/bwaki/catkin_ws/devel/.private/industrial_extrinsic_cal/lib/industrial_extrinsic_cal/rangeNmono] Error 1
make[1]: *** [CMakeFiles/rangeNmono.dir/all] Error 2
make: *** [all] Error 2

```

Bwaki, 2018-05-25:
Nobody know ?

drchrislewis, 2018-05-25:
I would have guessed it was a yaml_cpp version issue too. What version of linux are you using?
There was a big change in yaml between 14.04 and 16.04, but the errors you get don't look like that. 
There are several cpp files that use yaml, not just mutable joint state publisher. Could you try removing it from CMakelist.txt and see if the error is isolated to that file. I can't reproduce your error so its hard to debug.

Bwaki, 2018-05-25:
Ok when i remove "mutable joint state publisher" it build perfectly i will now try to change the version of yaml-cpp

Bwaki, 2018-05-25:
Yes the yaml-cpp version was the problem i just take the last version of it rebuild my WS and no error 
thanks you very much @drchrislewis

closed 2018-05-25 by Bwaki