ros-industrial/industrial_calibration #129: Binocular Calibration Disagrees With Reality

TomShkurti, 2018-10-01:
We've been working on adapting the ros-industrial extrinsic calibration utility to the binocular endoscope on a "da Vinci" surgical robot- specifically, we are determining the position of each camera in the endoscope with respect to an external target and using those transforms to find the camera-to-camera transformation. The solver usually converges within a few iterations to a very low error (~0.05), but the produced transformation is not consistent with the physical structure of the endoscope- the two camera lenses can be determined to be located ~5mm apart along their shared X-axis just using a ruler, but the calibration consistently places them ~3mm apart along the X AND Y axes.

I've attached the relevant configuration files. The only thing I can think of is that this is because we are using only one scene, but as setting up another will be a significant time investment and so I would like some confirmation that this is likely the issue and it is not just some other issue or a bug in the program itself.

[extrensic_yaml.zip](https://github.com/ros-industrial/industrial_calibration/files/2435693/extrensic_yaml.zip)

drchrislewis, 2018-10-02:
I believe the default initial condition assumes the large dot of the target
is in the lower left corner of the image.
Also, are you using the rail or the robot version? Also, did you go to
drchrislewis at github to get the after_intern branch?
Its the one with all the changes. I tried to get a pull request in before
the conference, but couldn't use my computer on the plane due to a lack of
room between the seats.
-Chirs

TomShkurti, 2018-10-03:
Took a look at your fork and the after_intern branch in particular; it's going to take some time (potentially a few days) to get it working with our system (which, while based on the rail calibration version, has been modified to accommodate a manually-actuated screw-driven stand underneath the endoscope which moves the target). The target does fill the image during calibration with the large circle at the lower left-hand corner- we have been using the intrinsic_calibration package also in ROS-i to get our intrinsics for each camera and been getting very good results with that, so at this point I'll just see what I can get from the forked version in terms of interocular calibration.

TomShkurti, 2018-10-04:
So after running a few other low-level checks we've noticed that there is a significant Y-discrepancy for easily-identifiable features between the *raw* images of the two cameras (X=158, Y = 329 versus X = 108 and Y = 348 for the intersection of two 1px-thick lines on a piece of paper). This suggests that the problem is not with the calibration package at all but rather it is reporting a (somewhat counter-intuitive) result accurately given what it is given- although it's not obvious from looking at the exterior of the device, the endoscope lenses are indeed offset in their shared Y-frame about half as much as in their X-frame, with negligible rotation.

So I think I am satisfied in closing this issue as something of a false alarm.

closed 2018-10-04 by TomShkurti