ros-industrial/industrial_calibration #135: advise regarding a suitable cost function

ManMan88, 2018-11-08:
First of all thank you for this great repository, really helpful.

I'm trying to calibrate a fixed camera to a robot with a target mounted on the robot, just like this example: [video](https://www.youtube.com/watch?v=AMMNXYtox_E).
I use the FixedCircleTargetCameraReprjErrorPK with one scene (and static target), and it works fine.
However, I want to increase the accuracy of the calibration, so I want to calibrate using several scenes.

I changed the static target to a moving target, and I tried several different cost functions, but it never seems to calibrate well. What cost function would you use for the case of a static camera and several scenes of a moving target on a robot (of course, when the images are taken, the target is static)?

Furthermore, in this [video](https://vimeo.com/106994711#t=965s), Chris notes that the middle of an ellipse in an image (of a projected circle) is not the center of the real circle. He says that this module has some cost functions that takes that into account. What cost functions are suitable for that?

Thanks a lot!

drchrislewis, 2018-11-08:
I've seen all your PRs and comments coming in. I have not had time to look at them, and I have Jury duty today. Hopefully I'll get a chance tomorrow. If you go to my fork of the rosi ical repo, you will find a branch called after_intern. Look at the new node called wrist_cal. It is a stand along node that will calibrate a camera to a robot if either the camera is on the robot or the target. There should even be a launch file example. I've got a few cleanup issues and some documentation before I do a PR.

drchrislewis, 2018-11-08:
@ManMan88 look at previous comment it was for you.

ManMan88, 2018-11-09:
Thanks for the quick reply, I'll try it next week.