ros-industrial/industrial_calibration #139: GPL license found?

130s, 2019-03-20:
Some packages in this repo depend on `ceres` (e.g. [industrial_extrinsic_cal](https://github.com/ros-industrial/industrial_calibration/blob/826b46b8fd297b1886b1efa94ed4c19ea4a1549c/industrial_extrinsic_cal/src/nodes/range_camera_excal.cpp#L40)) and [rgbd_depth_correction](https://github.com/ros-industrial/industrial_calibration/blob/826b46b8fd297b1886b1efa94ed4c19ea4a1549c/rgbd_depth_correction/include/depth_calibration/depth_calibration.h#L39), then statically link to it.

[ceres-solver.org](http://ceres-solver.org/installation.html) reads:

> By default, Ceres will link to `SuiteSparse` if it and all of its dependencies are present. Turn this OFF to build Ceres without SuiteSparse. `SuiteSparse` is licensed under a mixture of GPL/LGPL/Commercial terms. Ceres requires some components that are only licensed under GPL/Commercial terms.

IANAL, this gives me an impression that all downstream software of these packages need to follow what GPL requires to do.

- Was there any discussion regarding this?
- Was there any solution? (I hope I'm just wrong.)

drchrislewis, 2019-03-20:
I was certainly unaware of this. I'll look into it. We don't like using
GPL. -Chris

gavanderhoorn, 2019-03-20:
@130s wrote:
> IANAL, this gives me an impression that all downstream software of these packages need to follow what GPL requires to do.

Only if the `libceres` that is being linked to makes use of SuiteSparse, as also stated in the quote you include from the Ceres installation manual.

I realise it's common to have Ceres link to SuiteSparse, but it's not required.

drchrislewis, 2019-03-21:
I believe the Ceres debian which includes a copywrite declaration is in
error because it include suitsparse which is GPL. It is possible to build
ceres with suitsparse disabled. I don't know what speed ramifications this
might cause. However, I'd recommend that if anyone delivers a system to a
commercial customer or creates a debian using ceres that they build it from
source before linking to it. That way suitsparse can be excluded.
Otherwise,the copywrite for the derived work would have to be GPL. Please
understand that I am not a lawyer, so my recommendations are just an
uneducated opinion.  I've only recently started using ceres as a debian.
Before, I'd always built from source. I guess I'll go back to that. -Chris