ros-industrial/industrial_calibration #215: Documentation Request: Clarify initial guess parameters for KinematicCalibrationProblemPose6D

fschiro, 2025-07-18:
The documentation for KinematicCalibrationProblemPose6D requires three initial guesses but doesn't clearly explain their relationship or how to derive them, particularly for common setups.

Setup: Static camera observing a robotic arm with an ArUco marker attached to the end effector (link6).
My interpretation of the required parameters:

1. camera_mount_to_camera_guess - Identity matrix for static camera
2. target_mount_to_target_guess - Pose of ArUco marker in link6 frame
3. camera_base_to_target_base_guess - Pose of robot base in camera frame

Issue: The third parameter (camera_base_to_target_base_guess) appears redundant given the provided dataset and other parameters.

Since we have:
* Target poses in camera frame (from observations dataset)
* target_mount_to_target_guess (marker → link6 transform)
* Robot kinematics (DH parameters)

We can derive camera_base_to_target_base_guess by: camera → target → link6 → base (via forward kinematics).

Questions:
1. Is my interpretation of these parameters correct?
2. If the third parameter is indeed derivable, why is it required separately?
3. For practical implementation, should users simply compute the mean robot base pose from their observation dataset as the initial guess?

fschiro, 2025-07-18:
I think the convention is x_to_y means y-pose in x-frame, and learning that cleared things up a lot.

marip8, 2025-07-18:
Let me first caveat this response by saying that the kinematic calibration capabilities in this repository are pretty experimental, so bear that in mind as you try to use them.

If you haven't already seen them, check out the [unit test](https://github.com/ros-industrial/industrial_calibration/blob/main/test/dh_chain_kinematic_measurement_utest.cpp) and [example](https://github.com/ros-industrial/industrial_calibration/blob/main/examples/kinematic_calibration.cpp) for examples of how to use this functionality.

## Questions 1, 2

> Questions:
> 
> 1. Is my interpretation of these parameters correct?
> 2. If the third parameter is indeed derivable, why is it required separately?

Your interpretation is mostly correct. This optimization is intended to support a couple use cases:

### Single robot, camera mounted to robot, static target
  * `camera_chain`
    * DH kinematic chain for the robot
  * `target_chain`
    * empty DH kinematic chain (i.e., identity transform)
  * `camera_mount_to_camera_guess`
    * transform between robot flange to camera optical frame
  * `target_mount_to_target_guess`
    * transform from target chain base frame to target origin frame.
  * `camera_base_to_target_base_guess`
    * Transform between camera chain (i.e., robot) and target chain (empty); typically identity transform

If `camera_base_to_target_base_guess` is set to identity, `target_mount_to_target` effectively represents the transform from the robot base to the static target frame.

In this case, `camera_base_to_target_base_guess` is redundant. You can use set the `masks` variable in such a way that the calibration will not attempt to optimize that transform.

```
// Mask the 3 variables associated with the camera base to target base position
prob.masks[6] = {1, 2, 3};
// Mask the 3 variables associated with the target mount to target base angle axis 
prob.masks[7] = {1, 2, 3};
```

### Single robot, target mounted to robot, static camera

Effectively the same as the single robot case above, but reverse the camera and target terminologies. `camera_base_to_target_base_guess` is still redundant and should be masked before running the optimization.

### Two robots, target mounted to one robot, camera mounted to other robot

Two things change from the above use-cases:
- The `target_chain` DH kinematic chain is no longer empty and represents the second robot.
- `camera_base_to_target_base_guess` represents the transform between the two robot base frames.

In most cases you won't perfectly know the `camera_base_to_target_base` transform (e.g., physical installation never matches CAD exactly), so it can be worthwhile to try to optimize it. If you feel like you know this transform well enough (e.g., from CAD or other measurements), you can still mask it to prevent it from being optimized during the calibration.

## Question 3

> Issue: The third parameter (camera_base_to_target_base_guess) appears redundant given the provided dataset and other parameters.

Yes, but only in your use-case. Set this to identity and mask it in the optimization as described above.

> Since we have:
>
> Target poses in camera frame (from observations dataset)
> target_mount_to_target_guess (marker → link6 transform)
> Robot kinematics (DH parameters)

`target_mount_to_target_guess` is not marker -> link6 transform. In your case `target_mount_to_target_guess` should be the transform from the robot base to where your static target is placed.

> 3. For practical implementation, should users simply compute the mean robot base pose from their observation dataset as the initial guess?

I'm not sure I understand this question. Effectively what the optimization is doing is comparing measurements from your sensor to your understanding of the rest of the geometry:

```
camera -> target (from sensor) = camera -> camera mount -> camera base (via camera DH chain) -> target base -> target mount (via target DH chain) -> target
```

The calibration optimizes the DH chain parameters and the 3 provided transforms (all subject to mask exclusions) such that the above transform path "equation" has minimal squared error. You don't need to do any sort of robot pose averaging. You just need to provide these initial guess transforms and joint state measurements and the calibration does the rest.

## Misc.

> The documentation for KinematicCalibrationProblemPose6D requires three initial guesses but doesn't clearly explain their relationship or how to derive them, particularly for common setups.

If you have suggestions for clarifying, feel free to submit a PR to improve the documentation or post your suggestions here and I can incorporate them. I can work on summarizing this discussion into documentation if you find this useful.